uji-ros-pkg/underwater_simulation #53: cannot install uwsim in ubuntu 14 ros indigo

hughhugh, 2016-04-23:
Hi, I choose the way of Source-based installation in Groovy and later distributions to install uwsim, and I have git clone all the source files, In my workspace, I compile source code using 
` rosdep install --from-paths src --ignore-src --rosdistro groovy -y'
' catkin_make_isolated --install`

the error occurs in the uwsim/src/ROSInterface.cpp.o, the reason is the robot_state_publisher :publishFixedTransforms(const std::string& tf_prefix) in ROSInterface.cpp is not agreed with the opt/ros/indigo/share/robot_state_publisher. 

When using ‘sudo apt-get install ros-indigo-robot-state-publisher‘, the build of uwsim fails again.
there are six source packages to be catkin_make-ed, only underwater_simlation can not be build successfully.
The problem is still in the robot_state_publisher. Thus, how should I configure robot_state_pulisher in ros to help catkin_make the underwater_simulation package. Thank you.

perezsolerj, 2016-04-25:
Hi,

I have just checked because robot_state_publisher released a new version this weekend and everything seems  to be working with the last version too.

I have seen you copied "rosdep install --from-paths src --ignore-src --rosdistro **groovy** -y" I guess you already did it but if you are in indigo (as title claims) you should change groovy for indigo.

Besides this robot_state_publisher changed its interface a few months ago, so probably your uwsim or robot_state_publisher versions are outdated. So try updating robot state publisher through "sudo apt-get update" and "sudo apt-get install ros-indigo-robot-state-publisher". Then update UWSim with "git fetch" and "git pull" in the src folder. Finally try a fresh build of UWSim (delete build_isolated, devel_isolated and install_isolated) to make sure it uses the new library.

perezsolerj, 2016-05-16:
closing due to inactivity

closed 2016-05-16 by perezsolerj