uji-ros-pkg/underwater_simulation #57: [kinetic-devel] Build has never succeeded on ROS buildfarm

130s, 2017-07-01:
(Maybe) since its release into ROS Kinetic https://github.com/ros/rosdistro/pull/13689, I'm afraid `uwsim` might have never been built on buildfarm. At least AFAI see, the last 30 trials all failed (link to Jenkins [for 64bit](http://build.ros.org/view/Kbin_uX64/job/Kbin_uX64__uwsim__ubuntu_xenial_amd64__binary/), [for 32b](http://build.ros.org/view/Kbin_uX32/job/Kbin_uX32__uwsim__ubuntu_xenial_i386__binary/)).

Looking at its build log, [e.g. this](http://build.ros.org/view/Kbin_uX64/job/Kbin_uX64__uwsim__ubuntu_xenial_amd64__binary/172/consoleFull), I found a lot of errors happening around `libxml++-2.6`, e.g.:
```
23:41:46 /usr/include/libxml++-2.6/libxml++/exceptions/validity_error.h:36:19: error: expected ‘;’ at end of member declaration
23:41:46    ~validity_error() noexcept override;
23:41:46                    ^
23:41:46 /usr/include/libxml++-2.6/libxml++/exceptions/validity_error.h:36:21: error: ‘noexcept’ does not name a type
23:41:46    ~validity_error() noexcept override;
23:41:46                      ^
23:41:46 /usr/include/libxml++-2.6/libxml++/exceptions/validity_error.h:36:21: note: C++11 ‘noexcept’ only available with -std=c++11 or -std=gnu++11
23:41:46 /usr/include/libxml++-2.6/libxml++/exceptions/validity_error.h:38:16: warning: override controls (override/final) only available with -std=c++11 or -std=gnu++11
23:41:46    void Raise() const override;
23:41:46                 ^
23:41:46 /usr/include/libxml++-2.6/libxml++/exceptions/validity_error.h:39:22: warning: override controls (override/final) only available with -std=c++11 or -std=gnu++11
:
23:41:46 /usr/include/libxml++-2.6/libxml++/schema.h:48:40: error: ‘nullptr’ was not declared in this scope
```

130s, 2017-07-01:
Since its first release into Kinetic https://github.com/ros/rosdistro/pull/13689, there have been some commits that seem to be intended to fix build issues on Kinetic.

So I think the solution might be to just make another release using current HEAD.

130s, 2017-07-01:
ROS [prerelease test with the HEAD of today for Kinetic](https://travis-ci.org/uji-ros-pkg/underwater_simulation/jobs/249110997) passes (run in PR https://github.com/uji-ros-pkg/underwater_simulation/pull/59), which indicates that the current head works and supports the idea in https://github.com/uji-ros-pkg/underwater_simulation/issues/57#issuecomment-312435026

perezsolerj, 2017-07-18:
The configuration that is building is set as required while the ones that are failing are allowed to fail, until someone solves the read only issue satisfactorily.

closed 2017-07-18 by perezsolerj