uji-ros-pkg/underwater_simulation #60: Structured Light Projector in Kinetic

bvibhav, 2017-07-14:
I have been trying to make a simulation of laser projection with ROS Kinetic and UWsim. But whenever I use the plugin for <structuredLightProjector> in my scene, the brightness of the whole scene goes super low and there is no sign of the projection. Any ideas?

perezsolerj, 2017-07-17:
Hi,

Laser projection requires shaders, make sure your graphic card drivers are properly installed and working. When a laser projector is added to a scene and shaders are not supported there are glitches with scene brightness.

bvibhav, 2017-07-17:
Thanks for the reply.

The shaders were enabled and working on rest of the scene, except laser. Had the same issue in two different PCs. I reinstalled the package using source based installation. Seems to working with that. No idea what happened with normal install.

closed 2017-07-17 by bvibhav