uji-ros-pkg/underwater_simulation #64: Package ros-kinetic-uwsim doesn't exist

thepirate16, 2017-12-25:
Hello,

When trying to get the package through _sudo apt-get install ros-kinetic-uwsim_ , it directly does not exist. The ones that exist are the following:
_ros-kinetic-uwsim-bullet_
_ros-kinetic-uwsim-osgbullet_
_ros-kinetic-uwsim-osgocean_
_ros-kinetic-uwsim-osgworks_

Any solution to this?

davidfornas, 2017-12-25:
In Kinetic you have to compile those dependencies from reps and install
UWSim in from source.

thepirate16, 2018-01-26:
Hi again. Now I downloaded successfully the files. Nevertheless, when trying to run
`$ rosrun uwsim uwsim`

It dowloads the files but it says:

_Starting UWSim...
[rosrun] Couldn't find executable named uwsim_binary below /home/daniel/catkin_ws/src/underwater_simulation/uwsim_

I followed the full installation from sources step by step.
Any thoughts?

thepirate16, 2018-01-26:
Ok I am going to start a new issue for this.

closed 2018-01-27 by daniglezad

x1aoo, 2019-02-20:
The following instructions are really helped me when I trying to install uwsim in kinetic

> Installation instructions here:
> 
> http://www.irs.uji.es/uwsim/wiki/index.php?title=Installing_UWSim