uji-ros-pkg/underwater_simulation #65: Could not find executable named uwsim_binary

thepirate16, 2018-01-26:
Hi again. Now I downloaded successfully the files. Nevertheless, when trying to run
`$ rosrun uwsim uwsim`

It dowloads the files but it says:

_Starting UWSim...
[rosrun] Couldn't find executable named uwsim_binary below /home/daniel/catkin_ws/src/underwater_simulation/uwsim_

I followed the full installation from sources step by step.

I tried to just copy the file it requests under the location where it looks for it, but has not worked. Then it says:

_Starting UWSim...
Loading SimulatedDevices plugin: 'DredgeTool_Factory'
terminate called after throwing an instance of 'pluginlib::LibraryLoadException'
  what():  Could not find library corresponding to plugin DredgeTool_Factory. Make sure the plugin description XML file has the correct name of the library and that the library actually exists.
/home/daniel/catkin_ws/src/underwater_simulation/uwsim/src/uwsim: line 23: 29256 Aborted                 (core dumped) rosrun uwsim uwsim_binary --dataPath ~/.uwsim/data $@_

Any thoughts?
Thank you.

davidfornas, 2018-01-27:
You have to do "catkin_make install" in your workspace and then "source
your_workspace/install/setup.sh", then "rosrun uwsim uwsim".

thepirate16, 2018-01-27:
**It works!**

Thank you very much! Gràcies :)

closed 2018-01-27 by daniglezad