uji-ros-pkg/underwater_simulation #67: Problems about range snesor

roaddy, 2018-03-15:
I think ,in default Tank Scene with G500, the virtual range sensor is mounted to Part0 with pitch -1.57 rad. So it is pointing down to measure the distance between ROV and the floor. But when I try to use range sensor as a forward looking sonar, I change the pitch to 0. However, the data from topic doesn't change at all. It seems that it still feedback the distance between ROV and floor. I wonder how to use range sonar as FLS?

jake3991, 2019-01-04:
Did you make any progress on this? I'm also looking to utilize this system with a forward looking imaging sonar.

roaddy, 2019-01-08:
> Did you make any progress on this? I'm also looking to utilize this system with a forward looking imaging sonar.

Sorry, I didn't work on it.

closed 2019-01-08 by roaddy