uji-ros-pkg/underwater_simulation #68: Install on Ubuntu 18 LTS with ROS melodic

DeniseBryson, 2018-08-29:
I'm trying to install uwsim on Ubuntu 18 LTS with the rosdistro = melodic.
I started with:
`wstool init src`

I edited the .rosinstall file as mentioned in the wiki and ran:
`wstool update`
It downloaded a lot but I got some errors as well which look like they are distro dependend. Rerunning the command gives me the following errors.

> [src/uwsim_osgocean] Updating /home/cooper/Programme/WaterGlider/catkin_ws/src/src/uwsim_osgocean
> WARNING [vcstools] Command failed: 'git checkout melodic-devel'
>  run at: '/home/cooper/Programme/WaterGlider/catkin_ws/src/src/uwsim_osgocean'
>  errcode: 1:
> error: pathspec 'melodic-devel' did not match any file(s) known to git.
> [/vcstools]
> [src/uwsim_osgworks] Updating /home/cooper/Programme/WaterGlider/catkin_ws/src/src/uwsim_osgworks
> WARNING [vcstools] Command failed: 'git checkout melodic-devel'
>  run at: '/home/cooper/Programme/WaterGlider/catkin_ws/src/src/uwsim_osgworks'
>  errcode: 1:
> error: pathspec 'melodic-devel' did not match any file(s) known to git.
> [/vcstools]
> [src/uwsim_bullet] Updating /home/cooper/Programme/WaterGlider/catkin_ws/src/src/uwsim_bullet
> WARNING [vcstools] Command failed: 'git checkout melodic-devel'
>  run at: '/home/cooper/Programme/WaterGlider/catkin_ws/src/src/uwsim_bullet'
>  errcode: 1:
> error: pathspec 'melodic-devel' did not match any file(s) known to git.
> [/vcstools]
> [src/uwsim_osgbullet] Updating /home/cooper/Programme/WaterGlider/catkin_ws/src/src/uwsim_osgbullet
> WARNING [vcstools] Command failed: 'git checkout melodic-devel'
>  run at: '/home/cooper/Programme/WaterGlider/catkin_ws/src/src/uwsim_osgbullet'
>  errcode: 1:
> error: pathspec 'melodic-devel' did not match any file(s) known to git.
> [/vcstools]
> [src/underwater_simulation] Updating /home/cooper/Programme/WaterGlider/catkin_ws/src/src/underwater_simulation
> WARNING [vcstools] Command failed: 'git checkout melodic-devel'
>  run at: '/home/cooper/Programme/WaterGlider/catkin_ws/src/src/underwater_simulation'
>  errcode: 1:
> error: pathspec 'melodic-devel' did not match any file(s) known to git.
> [/vcstools]
> Exception caught during install: Error processing 'src/uwsim_osgocean' : [src/uwsim_osgocean] Update Failed of /home/cooper/Programme/WaterGlider/catkin_ws/src/src/uwsim_osgocean
> Error processing 'src/uwsim_osgworks' : [src/uwsim_osgworks] Update Failed of /home/cooper/Programme/WaterGlider/catkin_ws/src/src/uwsim_osgworks
> Error processing 'src/uwsim_bullet' : [src/uwsim_bullet] Update Failed of /home/cooper/Programme/WaterGlider/catkin_ws/src/src/uwsim_bullet
> Error processing 'src/uwsim_osgbullet' : [src/uwsim_osgbullet] Update Failed of /home/cooper/Programme/WaterGlider/catkin_ws/src/src/uwsim_osgbullet
> Error processing 'src/underwater_simulation' : [src/underwater_simulation] Update Failed of /home/cooper/Programme/WaterGlider/catkin_ws/src/src/underwater_simulation
> 
> ERROR in config: Error processing 'src/uwsim_osgocean' : [src/uwsim_osgocean] Update Failed of /home/cooper/Programme/WaterGlider/catkin_ws/src/src/uwsim_osgocean
> Error processing 'src/uwsim_osgworks' : [src/uwsim_osgworks] Update Failed of /home/cooper/Programme/WaterGlider/catkin_ws/src/src/uwsim_osgworks
> Error processing 'src/uwsim_bullet' : [src/uwsim_bullet] Update Failed of /home/cooper/Programme/WaterGlider/catkin_ws/src/src/uwsim_bullet
> Error processing 'src/uwsim_osgbullet' : [src/uwsim_osgbullet] Update Failed of /home/cooper/Programme/WaterGlider/catkin_ws/src/src/uwsim_osgbullet
> Error processing 'src/underwater_simulation' : [src/underwater_simulation] Update Failed of /home/cooper/Programme/WaterGlider/catkin_ws/src/src/underwater_simulation

After that I tried to continue anyway. Which is stupid probably.

I ran:
` rosdep install --from-paths src --ignore-src --rosdistro melodic -y -r`
Getting this error besides lots of installed dependecies.

> ERROR: the following packages/stacks could not have their rosdep keys resolved
> to system dependencies:
> uwsim_osgocean: No definition of [catkin] for OS version [artful]
> underwater_vehicle_dynamics: No definition of [tf_conversions] for OS version [artful]
> underwater_sensor_msgs: No definition of [visualization_msgs] for OS version [artful]
> uwsim_bullet: No definition of [catkin] for OS version [artful]
> uwsim: Cannot locate rosdep definition for [xacro]
> uwsim_osgbullet: No definition of [uwsim_osgworks] for OS version [artful]
> uwsim_osgworks: No definition of [catkin] for OS version [artful]
> Continuing to install resolvable dependencies...
> #All required rosdeps installed successfully
> 

and then:
` catkin_make_isolated`
This installs 6 packages but fails to install uwsim with the error:

> CMake Error at CMakeLists.txt:24 (find_package):
>   By not providing "Findosg_markers.cmake" in CMAKE_MODULE_PATH this project
>   has asked CMake to find a package configuration file provided by
>   "osg_markers", but CMake did not find one.
> 
>   Could not find a package configuration file provided by "osg_markers" with
>   any of the following names:
> 
>     osg_markersConfig.cmake
>     osg_markers-config.cmake
> 
>   Add the installation prefix of "osg_markers" to CMAKE_PREFIX_PATH or set
>   "osg_markers_DIR" to a directory containing one of the above files.  If
>   "osg_markers" provides a separate development package or SDK, be sure it
>   has been installed.
> 
> 
> -- Configuring incomplete, errors occurred!
> See also "/home/cooper/Programme/WaterGlider/catkin_ws/build_isolated/uwsim/CMakeFiles/CMakeOutput.log".
> See also "/home/cooper/Programme/WaterGlider/catkin_ws/build_isolated/uwsim/CMakeFiles/CMakeError.log".
> <== Failed to process package 'uwsim': 
>   Command '['/home/cooper/Programme/WaterGlider/catkin_ws/devel_isolated/uwsim_osgbullet/env.sh', 'cmake', '/home/cooper/Programme/WaterGlider/catkin_ws/src/src/underwater_simulation/uwsim', '-DCATKIN_DEVEL_PREFIX=/home/cooper/Programme/WaterGlider/catkin_ws/devel_isolated/uwsim', '-DCMAKE_INSTALL_PREFIX=/home/cooper/Programme/WaterGlider/catkin_ws/install_isolated', '-G', 'Unix Makefiles']' returned non-zero exit status 1
> 
> Reproduce this error by running:
> ==> cd /home/cooper/Programme/WaterGlider/catkin_ws/build_isolated/uwsim && /home/cooper/Programme/WaterGlider/catkin_ws/devel_isolated/uwsim_osgbullet/env.sh cmake /home/cooper/Programme/WaterGlider/catkin_ws/src/src/underwater_simulation/uwsim -DCATKIN_DEVEL_PREFIX=/home/cooper/Programme/WaterGlider/catkin_ws/devel_isolated/uwsim -DCMAKE_INSTALL_PREFIX=/home/cooper/Programme/WaterGlider/catkin_ws/install_isolated -G 'Unix Makefiles'
> 
> Command failed, exiting.

Is it possible to install uwsim with ros=melodic or is it impossible right now? 
Thanks in advance!

Lukas

davidfornas, 2018-08-29:
Melodic it is still not supported so the first failures are related with
the fact that melodic-devel branch does not exist yet in the repository.

DeniseBryson, 2018-09-07:
Stupid of me thanks for that. 
I switched to Ubuntu 16.04 now

closed 2018-09-07 by DeniseBryson