uji-ros-pkg/underwater_simulation #70: Melodic support plan?

ne-v0y, 2019-03-23:
Just wondering if this has been planned.

dcentelles, 2020-03-08:
Sorry for the late reply. UWsim can already be installed in melodic.

NicolayP, 2020-06-18:
Hi, i was trying to install UWsim from pacakges for melodci. When running 
`rosrun uwsim uwsim`
The data directory needs to be filled however, the ressources are not found (404 response):

The UWSim data directory (~/.uwsim/data) does not exist. We need to download ~300Mb of data and place it under ~/.uwsim/data. Note this is required to run UWSim.
Continue (Y/n) ?
Y
--2020-06-18 11:28:06--  http://www.irs.uji.es/uwsim/files/data.tgz
Resolving www.irs.uji.es (www.irs.uji.es)... 150.128.97.138
Connecting to www.irs.uji.es (www.irs.uji.es)|150.128.97.138|:80... connected.
HTTP request sent, awaiting response... 404 Not Found
2020-06-18 11:28:07 ERROR 404: Not Found.

Where can I find those ressources ?

dcentelles, 2020-06-18:
Sorry. We have had some problems on the server. It should work now. Please try iagain and let us know if you have any problems.