uji-ros-pkg/underwater_simulation #71: ~/.uwsim/data download fails

djoshuadulle, 2019-07-17:
Attempting to run on Ubuntu 16 with ROS Kinetic. Was able to build using release candidates and underwater_simulation repo.

Finally trying to rosrun uwsim uwsim and receiving the error:

`` --2019-07-17 18:37:47--  http://www.irs.uji.es/uwsim/files/data.tgz
Resolving www.irs.uji.es (www.irs.uji.es)... 150.128.97.138
Connecting to www.irs.uji.es (www.irs.uji.es)|150.128.97.138|:80... connected.
HTTP request sent, awaiting response... 404 Not Found
2019-07-17 18:37:47 ERROR 404: Not Found.

Starting UWSim...
Loading SimulatedDevices plugin: 'DredgeTool_Factory'
Loading SimulatedDevices plugin: 'ForceSensor_Factory'
Loading SimulatedDevices plugin: 'SimDev_Echo_Factory'
  . Setting localized world: 7.8e-05s
Loading URDF robot... 
  · robot/GIRONA500/g500_March11.osg: Error retrieving file robot/GIRONA500/g500_March11.osg Check URDF file or set your data path with the --dataPath option.
terminate called after throwing an instance of 'pluginlib::LibraryUnloadException'
  what():  According to the loaded plugin descriptions the class ForceSensor_Factory with base class type uwsim::SimulatedDeviceFactory does not exist. Declared types are  DredgeTool_Factory ForceSensor_Factory SimDev_Echo_Factory
/home/jdulle/catkin_ws/install_isolated/lib/uwsim/uwsim: line 23: 17680 Aborted                 (core dumped) rosrun uwsim uwsim_binary --dataPath ~/.uwsim/data $@
''

My guess is that http://www.irs.uji.es/uwsim/files/data.tgz no longer exists, so the follow error is caused by not having the ~/.uwsim/data file.

Is there a work around for this?  

Thanks!

djoshuadulle, 2019-07-18:
I used the prebuilt VM and took out the ~/.uwsim/data folders. They seem to not contain GIRONA500.

I have also read a lot of posts about sourcing ~/catkin_ws/install/setup.sh or setup.bash but that is not working for me either.

ne-v0y, 2019-07-18:
> Resolving www.irs.uji.es (www.irs.uji.es)... 150.128.97.138
Connecting to www.irs.uji.es (www.irs.uji.es)|150.128.97.138|:80... connected.
HTTP request sent, awaiting response... 404 Not Found
2019-07-17 18:37:47 ERROR 404: Not Found.

Seems like a resource issue

dcentelles, 2019-07-19:
Hi,

   some files have been removed from our server. We are in the process of
making them back available. The files needed to run the uwsim the first
time have been recovered but some of the newer scenes (pipes_scenes,
pipeFollowing, etc) present more problems. We expect that they will be
available next week.
   We will notify you as soon the issue has been completely solved. If you
need a specific scene, please, tell us and we will try to make it available
sooner.

   Sorry for the inconvenience,

   UWSim team

Missatge de Noni Hua <notifications@github.com> del dia dj., 18 de jul.
2019 a les 22:21:

> Resolving www.irs.uji.es (www.irs.uji.es)... 150.128.97.138
> Connecting to www.irs.uji.es (www.irs.uji.es)|150.128.97.138|:80
> <http://www.irs.uji.es)%7C150.128.97.138%7C:80>... connected.
> HTTP request sent, awaiting response... 404 Not Found
> 2019-07-17 18:37:47 ERROR 404: Not Found.
>
> Seems like a resource issue

ccs-ros, 2019-09-22:
Hi,i want to run on ubuntu16.04 with ros-kinetic,and i meet some error,just like these:

/home/ccs/catkin_uwsimulation/install_isolated/lib/libosg_markers.so: undefined reference to `osg::Geometry::setColorArray(osg::Array*)'
collect2: error: ld returned 1 exit status
CMakeFiles/osg_interactive_markers_demo.dir/build.make:134: recipe for target '/home/ccs/catkin_uwsimulation/devel_isolated/osg_interactive_markers/lib/osg_interactive_markers/osg_interactive_markers_demo' failed
make[2]: *** [/home/ccs/catkin_uwsimulation/devel_isolated/osg_interactive_markers/lib/osg_interactive_markers/osg_interactive_markers_demo] Error 1
CMakeFiles/Makefile2:67: recipe for target 'CMakeFiles/osg_interactive_markers_demo.dir/all' failed
make[1]: *** [CMakeFiles/osg_interactive_markers_demo.dir/all] Error 2
Makefile:138: recipe for target 'all' failed
make: *** [all] Error 2
<== Failed to process package 'osg_interactive_markers': 
  Command '['/home/ccs/catkin_uwsimulation/install_isolated/env.sh', 'make', '-j2']' returned non-zero exit status 2

Reproduce this error by running:
==> cd /home/ccs/catkin_uwsimulation/build_isolated/osg_interactive_markers && /home/ccs/catkin_uwsimulation/install_isolated/env.sh make -j2

can you help me solve this problem? which is the version of your openscenegraph? It looks like that the osg had some problem.thank you very much!

ccs-ros, 2019-09-22:
@djoshuadulle

djoshuadulle, 2019-09-23:
If I recall correctly, the way I got passed this step was by cloning the osg-release versions from the uji-ros-pkg repo. That solved the error, but led to several others. In the end, I was never able to run uwsim on Ubuntu 16.

closed 2019-09-23 by djoshuadulle